paulhayes/web-3d-model-customiser #9: Brim ears on the

james-fry, 2020-04-04:
Is it possible to add an option for mouse ears on the ends?
e.g.
[image]

Of course can be added in cad or slicer but it might help...

sebleedelisle, 2020-04-04:
Hey James, is this to help with bed adhesion? Have you found it necessary? I must admit I haven't seen anyone else mention it but if it turns out to be popular I would consider it.

james-fry, 2020-04-04:
Hi Yes - to prevent the print lifting from the the bed on the corners. Like a mini brim.
There are a number of people recommending it on the 3dcrowd.uk slack.
Since I have issues with this type of lifting I added on my prints and its working well.

[image]

sebleedelisle, 2020-04-04:
Good to know. I’ll definitely consider adding it 👍

sebleedelisle, 2020-04-06:
Done 😊 #14

closed 2020-04-06 by sebleedelisle

james-fry, 2020-04-06:
Thanks Seb - it works great!