paulhayes/web-3d-model-customiser #15: Option for original design

patriot1889, 2020-04-06:
Any chance of an option to use the original file? (without “PRUSA” and “RC3" moved)

For example, if you didn’t add any text at all the original design would be used. As soon as you add text, it goes to the version with the text where it is currently?

Alternatively, it could be a version with "PRUSA" on one side and "RC3" on the other, rather than "RC3" being on the inside. I find it strange that it was put on the inside, particularly when there is so much information around regarding comfort with prolonged usage (eg. NHS) - wouldn't the best idea be to have the inside as smooth and uniform as possible?

With this in place, the tool could be used to just make stacks larger than 4, if the user didn't want to add extra text.

sebleedelisle, 2020-04-07:
Hi there, thanks for the feedback! Can I ask if there's a particular reason behind this request?

patriot1889, 2020-04-07:
With speed being the most important thing right now in producing these (without sacrificing important quality), I find that the versions produced with this don't come out well around the small lettering.

Furthermore, without any additional modifications, they take longer to print than the original model. (15% slower with my current profile).

sebleedelisle, 2020-04-07:
So you want an entirely blank model, without even the small Prusa logo? Is that more important than having the original model with the full size PRUSA logo?

patriot1889, 2020-04-07:
Quite possibly. I suppose it is up for debate on how far to take it. Is the PRUSA logo important? Does it show that it is a certain standardised design? Possibly, but unlikely that it will actually matter. It seems a shame to remove the logo seeing as PRUSA put so much effort into this, for the greater good. Is 6% more printing time a problem, bearing it mind it will use more filament...

I am not suggesting I have the answers, I suppose it depends on each persons set up. My current print time for one is around 1 hour 25mins with the original stl, the PRUSA logo comes out fine, RC3 (which is smaller text) is a little rough, which is a shame as it is on the inside. That would suggest to me that the size of "PRUSA" is ideal and can be printed at speed. It seems that the more text that is added, the print time goes up. So my thinking is that it is probably more important to have "PLA"/"PETG" embossed on the side than "PRUSA", so that users are aware of the properties (and sufficient possible cleaning techniques) for each item.

PS. This project is great, keep up the good work.