paulhayes/web-3d-model-customiser #16: Option to increase spacing between bands

billt79, 2020-04-06:
This would be really useful for people wanting to print multiples with a larger nozzle diameter

owennewo, 2020-04-08:
+1, printing a stack with layer height of 0.4 (0.6 nozzle) ends up with fused bands

Vegsaad, 2020-04-10:
I have also had problems with welding when using larger nozzle sizes. Perhaps if the nozzle size or gap size was an input this would be useful for those trying to print as fast as possible overnight.

sebleedelisle, 2020-04-11:
Currently being tested in the dev deployment : http://modsebleedev.netlify.com - can you test and let me know how it works?

Vegsaad, 2020-04-11:
Hi Seb,

 

Thanks for your update I will test asap but at the moment I am out of filament – just finished 60+ face shields may get a resupply on Monday.

Was getting up in the night to n alarm to reset my bed so this will be a game changer for me.

 

Thanks again

David Glass

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

owennewo, 2020-04-11:
I'm trying your tool with a 2 stack using 0.5 nozzle and 0.4mm layer height.   From my limited Cura experience, the stl looked good when I examined the intersection, so I have some confidence that it should separate.  It's possible that I might run into problems unrelated to your STL (thermal creep) as my ender3 pro doesn't enjoy printing out this fast!  
FYI My last dual was 5hrs 15mins (0.5/0.25) and this is 3hrs 30min (0.5/0.4mm).  
Longer term I'd probably slow it down to 0.5/0.35 to give me a quad in 8 hours whilst I sleep but I'm not sure whether it's best for layer height to be a nice multiple of 20mm.

sebleedelisle, 2020-04-11:
> not sure whether it's best for layer height to be a nice multiple of 20mm.

With the new tool, the face shield height is adjusted to be a perfect multiple of whatever layer height you select, so you should be fine whatever you choose. But let me know!

sebleedelisle, 2020-04-11:
I'm gonna close this for now, let me know if you have trouble and I'll reopen it.

closed 2020-04-11 by sebleedelisle

owennewo, 2020-04-11:
Sad face. This was 0.5/0.4mm.
[image]

owennewo, 2020-04-11:
I think there is a dot missing in the model.  On the prusa stack there is a second dot in the center of the attached image (Prusa has 8 dots, yours has 7).  Yours is missing that dot.
[image]

sebleedelisle, 2020-04-11:
I did add these in a recent update but they maybe too small to come out at large layer heights. Can you check again and let me know?

sebleedelisle, 2020-04-12:
Also, I only count 5 "dots" on both the Prusa model and also ours...

owennewo, 2020-04-12:
There are 8 dots on prusa.  Attached are two prusa and two seblee images.  prusa_left_dots one has three dots, whereas seblee_central_dots has two (missing top left).  prusa_central_dots has two dots but seblee_central has one dot (missing central bottom).  Haven't bothered with right images as its just a mirror of left
[image]
[image]
[image]
[image]

sebleedelisle, 2020-04-12:
Got it, thanks 😊