paulhayes/web-3d-model-customiser #17: Please can you remove all text except RC3

IamGPIO, 2020-04-06:
Hey, I know it's ideal to have date, name, prusa, etc however in interest of strength and speed, can you have the option to omit this all and just leave the version number?

Thanks
Duncan

krisnoble, 2020-04-07:
You can already untick the boxes for material and date + leave the name blank. Think they updated this yesterday so make sure you're not using a cached version by doing a hard refresh (Usually Ctrl-F5 on Windows or Cmd-Shift-R on a Mac). I would argue keeping "Prusa" is valuable as well because that's the name of the design, but the text on the ones this tool outputs is smaller and less intrusive than the default Prusa models anyway.

IamGPIO, 2020-04-07:
Understood, thanks!

-- 
Duncan Coombe
Director
 +44 208 819 8527 <+44+208+819+8527>
 +44 7782 136346 <+44+7782+136346>
 duncan.coombe@dclx.co.uk
 WWW.DCLX.CO.UK
 Unit 24 Gardner Industrial Estate, Kent House Lane, Beckenham, BR3 1QZ
<https://maps.google.com/?q=Unit+24+Gardner+Industrial+Estate%2C+Kent+House+Lane%2C+Beckenham%2C+BR3+1QZ>
<http://www.dclx.co.uk> <https://www.facebook.com/dclxlighting>
<https://www.instagram.com/dclx_lighting/>
<https://twitter.com/dclx_lighting>