paulhayes/web-3d-model-customiser #19: Sacrificial layer for bed adhesion / sterility

RikAllen, 2020-04-07:
Hi. Might be a possibility to have a thin layer at the bottom that separates from the stack and is discarded to give better bed adhesion and no contamination from the bed. A built in raft.

sebleedelisle, 2020-04-07:
Hey Rik, is there any reason you can't have the slicer make a raft / brim? I would prefer to avoid features that are common in slicers if possible.

RikAllen, 2020-04-07:
Can achieve similar - brims leave a fair bit of cleanup. I'm finding that even without a brim I get a sharp edge on the bottom piece that would not be pleasant to wear. Either that or I set the bed lower and risk the whole stack failing. Raft might work better I guess - was just thinking in terms of something that was bespoke for the model rather than the slicer trying to bodge it. I might generate an extra mask at the bottom and cut almost all off with the slicer except for 2 or 3 layers.

Matt2k34, 2020-04-07:
Hey Rik, does it happen if you use the single stl on it's own? if you have a heated bed i'm wondering if the weight of the others is slightly melting the material at the base.

krisnoble, 2020-04-07:
Yeah I think Matt's right, sounds like "elephant footing" - have a play with "Initial Layer Horizontal Expansion" in Cura (not sure what it's called in other slicers sorry) and you'll want to set a negative value. I use -0.25mm for mine but YMMV

RikAllen, 2020-04-07:
Cheers - I'm on cura, no hotbed. I think the problem might be im squishing the initial layer too much in an attempt to get decent bed adhesion. I seem to be getting past that now, but the nights prints came off and were ruined.
Closing this as it can be dealt with in other ways.

closed 2020-04-07 by RikAllen