paulhayes/web-3d-model-customiser #23: Problem with infill using S3D to slice

rsmith21, 2020-04-07:
I added my details and downloaded the file without problems,but when I slice it the print preview shows infill between the visor support and head band.

sebleedelisle, 2020-04-07:
Hey Bob, not quite sure what you mean, what do you mean by "visor support"?  - can you do a screen grab or something?

rsmith21, 2020-04-07:
[image]
Also if I use Cura it tells me the file has holes, so I use the mesh tool to repair, but it still shows red for an error.

sebleedelisle, 2020-04-07:
Huh I never saw that before. Has that just started happening or is this the first time you tried it? I just pushed an update an hour or so ago.

sebleedelisle, 2020-04-07:
Also, can you let me know what settings you used, we'll try to replicate the error. Also, send me the STL - thanks!

rsmith21, 2020-04-07:
I don't think its new, I tried it when you first made it available, when I saw the problem I went back to standard printing to get the job done.

rsmith21, 2020-04-07:
The file you base the headband on, is that an STL , or the Prusa one.

sebleedelisle, 2020-04-07:
it's based on the Prusa STP file

rsmith21, 2020-04-07:
That is my problem, when I use that it also looks wrong, but if I use their STL it slices perfectly.Sorry to give you unnecessary grief, I should have realised.

sebleedelisle, 2020-04-07:
Which version of Cura? I'll download it and take a look. It'd be good if we can get it sorted!

rsmith21, 2020-04-07:
4.5 , would it be possible to base the system on the STL, or an option to use STP or STL.

sebleedelisle, 2020-04-07:
I'll look into it - thanks

sebleedelisle, 2020-04-07:
Just tried in Cura 4.5, don't seem to have the problem... could it be some kind of raft/brim setting? 
[image]

rsmith21, 2020-04-07:
I'll load the STP and see how it shows on mine, I normally only use the STL since my printers are not Prusa.

sebleedelisle, 2020-04-07:
the image above is from an STL created by our system

Matt2k34, 2020-04-07:
This happens to me if I use the STL from the prusa site within cura. It does NOT happen for me for the STLs generated by the customizer.

@rsmith21 can you confirm you're using the customised (auto dated) STL?

rsmith21, 2020-04-07:
[image]
Cura didn't open the STP file for me, I opened the 3MF file and this is what I get, in a sliced preview.

Matt2k34, 2020-04-07:
The customizer doesn't generate a 3mf. That is from the prusa site (which is broken in cura)

sebleedelisle, 2020-04-07:
I'm also seeing the issue when loading the Prusa 3mf file. @rsmith21 can you send me your STL that your getting from our system that's causing the problem? Thanks!

rsmith21, 2020-04-07:
I tried to drag and drop the file but get the error " we don't support that kind of file"

rsmith21, 2020-04-07:
I'm sure that if no one else is having a problem slicing the file its a setting I have, please don't waste any time on it. I can get by for now without embossing or use the 3D builder method. 
Thanks for looking at it, will I just close it.

sebleedelisle, 2020-04-07:
I'm wondering if it's a bug that only appears with your particular choice of name or quality. Does it still happen no wonder how you customise the file?

closed 2020-04-07 by rsmith21