paulhayes/web-3d-model-customiser #24: Increase font size to accommodate larger nozzles

romikk, 2020-04-08:
With 0.8 nozzle and 0.64 layer size, the letters become illegible.

[image]

krisnoble, 2020-04-08:
Could be worth incorporating this with #16 - if you can let us specify our layer height then the gap between bands needs to be the layer height, and the text size could be calculated based off it too?

james-fry, 2020-04-10:
I have started double spacing my name.
e.g. "J . F R Y" which reduces problems with legibility due to "ringing" noise since print speeds are ramped right to max.

paulhayes, 2020-04-11:
I'm going to investigate this once we've added nozzle/layer height options.

sebleedelisle, 2020-04-11:
I'm happy to take this one - it's more in my area I think