paulhayes/web-3d-model-customiser #28: Extra skin being generated

DJDC13, 2020-04-09:
I am using latest Cura and printing on an Ender 3. Slicing a single one works great and all the options generate a good .stl file but when I try to generate a stack my slice is generating a skin between the bands? This seems to happen on any stack size except for a single. I am using 0.6mm nozzle and have a 0.4mm layer height.

sebleedelisle, 2020-04-09:
We can't seem to reproduce this issue, can you try the latest update to the site, and if it still happens, screenshot the settings on the customiser page and post here, thanks!

sebleedelisle, 2020-04-11:
OK I've found this issue occurring with layer heights of more than 0.25mm. I've just made a new version of the site for testing different layer heights, can you try it please? https://modsebleedev.netlify.com

DJDC13, 2020-04-12:
single ones work on this new model but I still cant get the stacked model to work unless i enable the 'keep disconnected faces' option. I have tried a few different layer heights and different stack heights.

DJDC13, 2020-04-15:
after a lot of trial and error I think my issue is being cause by the date stamp. When I remove the date it slices perfectly every time. With the date there is sometimes 1 layer sometimes a couple that are slightly shifted and Cura is picking this up and adding the skin at that layer