paulhayes/web-3d-model-customiser #30: Gap too small for larger nozzles causing welding - input gap size

Vegsaad, 2020-04-10:
Most people are printing with larger nozzles to increase print speed and productivity. When I tried a two-stack with a 0.8mm they welded together and could not be separated without scrapping both parts. Adding an option to increase the gap by either directly entering the Gap size or indirectly by entering the nozzle size would solve this and allow me to print fast and in numbers overnight.

sebleedelisle, 2020-04-11:
Hi there, I'm currently testing the layer height setting, can you give it a try and report back? https://modsebleedev.netlify.com

sebleedelisle, 2020-04-11:
Closing this for now, let me know how it works for you.

closed 2020-04-11 by sebleedelisle

Vegsaad, 2020-04-11:
Will do thanks.

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com