paulhayes/web-3d-model-customiser #32: Store settings?

krisnoble, 2020-04-10:
Just an idea for a QOL improvement really, would it be possible to have an option to store our settings in a cookie or some other local method so that the tool just automatically generates what we had last time?

I know there's GDPR to think about with names but if it was worded as something like "store my settings in a cookie (optional)" then I think that would be sufficient for informed consent?

I've tried to figure out how to do it myself with a view to making a PR but struggling, my brain is running suboptimally right now :)

paulhayes, 2020-04-11:
Thanks for the suggestions. I'll implement today.

paulhayes, 2020-04-11:
Fixed in 4a6558a3b31f0a989ac893aeb709922e27fa0373

closed 2020-04-11 by paulhayes