paulhayes/web-3d-model-customiser #33: ears don't print

lobodelmar, 2020-04-10:
I'm using Ultimaker 2+ extended. The ears are very useful for stability and even though I choose them and I can see them on Cura:
https://photos.app.goo.gl/xN7fsKAmPRe36XZA8
they don't show up in the preview tab of Cura:
https://photos.app.goo.gl/9DbN64kXLWchb6b46
Nor do they print:
https://photos.app.goo.gl/UWgJNetFoCnTN2HA6
Any ideas why?

Thanks Seb for such a great tool! Makes printing these masks so much faster! Great contribution to the effort.

sebleedelisle, 2020-04-10:
Hi there, what layer height are you using? Is it the same when using a layer height of 0.25mm?

lobodelmar, 2020-04-10:
0.5 because I downloaded a Cura settings file from the 3DCrowd slack channel from someone that said it worked well like that.

I notice now that it has 2 settings. Layer height + initial layer height. Changing layer height made no difference. Changing initial layer height to 2.5 made the ears appear in the preview. I'll go ahead and print and report back if that fixed it.

Thanks Seb!

sebleedelisle, 2020-04-10:
Thanks for that, yeah at the moment the stacks need to have a layer height of 0.25mm just like the Prusa stacks. I'm looking into customising this (as MANY have requested) but right it's best to use 0.25mm.

sebleedelisle, 2020-04-10:
I've reopened this issue and renamed it to remind me to look at this once I add customisable layer heights.

lobodelmar, 2020-04-10:
Yep, it worked in print too. I tried .25 for first layer as you suggested but then switched to .4 because it also worked.

Thanks for the tip all good :).

JW6677, 2020-04-16:
A layer hight text box would be really useful. I've tried downloading the branch with this using npm init and npm install web-3d-model-customiser  but I got a 404. Could you try and get this update on the website soon because I'm trying to use a 0.8mm nozzle with a 0.6 mm layer height and it isn't printing tabs at all and fuses the stack together.

sebleedelisle, 2020-04-16:
The adjustable layer height branch can be viewed online at http://modsebleedev.netlify.com but this doesn't alter the the mouse ear heights yet