paulhayes/web-3d-model-customiser #34: outside pins are stuck to support wall in mulstistack

ales1, 2020-04-10:
One more suggestion please, in stacked version there is extra support printed on the front for the pins. I have a problem removing the support on external pins, middle two pins seems to only touch the base, but side pins do touch support walls on the side and stick to them pretty well, expanding those by 1mm would solve the problem

sebleedelisle, 2020-04-10:
What layer height and extrusion width are you printing in please?

ales1, 2020-04-10:
Hey, layer height is 0.3. Extrusion width is a default 0.4 mounted on flsun qq s.

paulhayes, 2020-04-10:
Ha we have the same printer!

ales1, 2020-04-10:
nice one:) there is a channel on slack printersettings-flsun-qq-s if you want to join.  this is my second week printing, never did that in a past. If you have any expierience please share. Have you managed to do reasonable prints in PETg?

twistedsanity, 2020-04-11:
I'm having the same issue, the outside pins are closer to the rear wall of the support (offset) and when running high speed with larger width (to reduce print time) it sticks. The middle 2 pins are fine as they are evenly spaced but the edge two aren't. I would love it if you were just able to tweak it slightly to even it up, and maybe add an extra 0.2. 
That being said, awesome work!!!! thank you!

sebleedelisle, 2020-04-11:
Hopefully fixed in the current dev version ( http://modsebleedev.netlify.com ) - can you try it out and let me know? You can check the layer height feature while you're there 😊

ales1, 2020-04-11:
Printing the last one in current stack, will try on next go

sebleedelisle, 2020-04-11:
Closing this for now, let me know if you still have trouble and I'll reopen it.

closed 2020-04-11 by sebleedelisle

ales1, 2020-04-11:
hey,
just finished a stack of 3, another small fix is needed.
Looking from the front (so you see all those pins in front of you). Left pin didn't have adhesion to the support walls anymore, two middle as good as in the past, but right one I would say even worse.
Exactly the same on both examples from the same stack, level 0 & 1

sebleedelisle, 2020-04-12:
Hi there, thanks for the feedback. Did you have "detect thin walls" switched on? 
What do you mean by pins? What do you mean by support walls? Sorry but it's hard to understand when there's no standardised terminology!

ales1, 2020-04-12:
Hey Seb, I use Cura and I didn't have 'think wall' option selected, will do another stack of three soon will confirm here. And will try to be more specific with description or even a picture. Thanks!