paulhayes/web-3d-model-customiser #36: Error in update on start

paulhayes, 2020-04-11:
Getting the following error in the console on start
```blob:https://mod.seb…-308520aec019:10317 Uncaught TypeError: Cannot read property 'scale' of undefined
    at main (blob:https://mod.seb…-308520aec019:10317)
    at blob:https://mod.seb…-308520aec019:10434
```

sebleedelisle, 2020-04-11:
I added a new object (cutouts) and tried to replicate the methodology that you set up for the extras. But maybe I messed up and the cutouts object hasn't been loaded properly the first time it tried to update the models? It seems to be fine on subsequent attempts.

sebleedelisle, 2020-04-11:
I fixed it by checking that the model is an object before trying to use it but it probably needs fixing properly 😊

closed 2020-04-11 by paulhayes