paulhayes/web-3d-model-customiser #39: Error in Model and Problem with Embedded Writing - FlashPrint

BlackMeister, 2020-04-11:
Hi, thanks for doing this.  Used a few days ago and all worked well.  Over the last couple of days I have noticed issues when I load the model into FlashPrint (for FlashForge) with the indent writing that was fine before, but now loads strangely and mainly fills in and has strange line artefacts over the wording when FlashPrint fixes the model. This happens after I get an "Error has occurred to the model", if I don't fix when I go to print you can see missing layers within the print. When I do fix it, it sorts the missing layers but messes up the writing.  Just loaded old models (which can't use as they have old dates embedded) and while it still asks to repair the model, the embedded writing is fine and the model prints fine.

sebleedelisle, 2020-04-11:
Can you change your layer height to 0.25mm and see if it still happens?

sebleedelisle, 2020-04-11:
Similar to #28

BlackMeister, 2020-04-11:
So it's strange.  Already set to 0.25.  However, after I select print to save the file, then I look at the pre print model, with 'PETG' on the right and say my name on the left.  On the left the name is still messed up / filled in.  On the right the bottom layer only shows the letter 'P' and the remain layers (selected 6) have 'ETG' with 'P' missing.  If I don't repair the model it's the same, lots of errors of missing layers.

BlackMeister, 2020-04-11:
It's your discussion forum, but not sure my issue is the same as #28.

sebleedelisle, 2020-04-11:
Sorry not sure what happened there.

BlackMeister, 2020-04-11:
The same problem of the error and the embedded writing occurs with 1 in the stack.

sebleedelisle, 2020-04-11:
Unfortunately there's an issue with the underlying OpenJsCad lib that produces non-manifold models, and I don't think FlashPrint can fix them. You can try fixing the models in Netfabb maybe? Or else send me an STL and I'll make sure that there's nothing else going on.

BlackMeister, 2020-04-11:
Thanks.  Tried playing around with PrusaSlicer, but still get some errors when back in FlashPrint and not sure FlashForge accepts G-Code.  If you could upload a reapired six stack with no date and no info that would be great.

sebleedelisle, 2020-04-11:
You can export an STL from PrusaSlicer once it's fixed, which you can then
load into FlashPrint

BlackMeister, 2020-04-11:
Did that, as an stl and seperatly an obj.  Nearly all repaired, but there was still a missing layer in the bottom mask.

sebleedelisle, 2020-04-11:
Yeah I'm seeing that sometimes too, but Netfabb seems to be able to fix
this error (although Paul is trying to get to the bottom of this on the
OpenJsCad side as we speak, so fingers crossed!)

BlackMeister, 2020-04-11:
Managed to get a 5 stack to work via Prusa, this is new to me so thanks for bringing that skill. I have a Mac so think Netfabb only MS?

sebleedelisle, 2020-04-11:
Ah that's cool 😊 I never downloaded the Netfabb app but there's an online
version you can try.

BlackMeister, 2020-04-11:
Just to let you know.  No matter what I do in repairing the file in Prusa, it all looks fine when I import it and is, except when I then slice/save the file in FlashPrint, it seems to add a raft even when a raft is not selected.