paulhayes/web-3d-model-customiser #43: Rounding errors causing model to be not exactly 20mm

pierreh, 2020-04-18:
there seems to be a slight discrepancy in the model height, when I measure in Fusion 360 I get 19.998. Which is causing the slicer to sometimes not print the last layer and you end up with a gap of twice the layer height
I’m guessing that this rounding issue can probably compound itself when adding more stacks or perhaps cause some gaps in the stack to be 1 layer and others 2 layers

**Seb Lee-Delisle - Brighton**
What layer height please? The model height is adjusted to be divisible by an exact number of layers

**Pierre Havelaar**
I know, I’ve seen the code, but there is always rounding errors, because numbers are floating point in Javascript and you are scaling the 3D model by the model height. You want get an exact number because of that. I’ve seen it for various layer heights: 0.2, 0.4, 0.5, 0.6

sebleedelisle, 2020-04-21:
I think I've fixed this now but perhaps you could test for me? It's currently deploying to http://modsebleedev.netlify.app @pierreh

pierreh, 2020-04-21:
@sebleedelisle it still looks the same to me, however it doesn't seem to cause problems for me at the moment. If you'd like you can close the issue.

pierreh, 2020-04-21:
Considering javascripts floating point calculations it might be impossible to fix.